patience4711/ESP32-read-APS-inverters #1: ESP32-ECU_v0-1a.bin boot loop on D1 Mini ESP32 WROOM

bs-github, 2023-04-12:
I've tried the ESP32-ECU_v0-1a.bin firmware on the D1 Mini ESP32 WROOM, but it goes right into a boot loop after flashing.

Can you please put the source code on here as well like on the original project, so we can debug on our own?

```
➜  esptool.py --port /dev/cu.usbserial-01A13DA0 --baud 460800 write_flash 0x0000 ~/Downloads/ESP32_ECU_v0_1/ESP32-ECU_v0-1a.bin   && grabserial -d /dev/cu.usbserial-01A13DA0 -b 115200
esptool.py v4.5.1
Serial port /dev/cu.usbserial-01A13DA0
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b4:e6:2d:e9:4e:dd
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x000fffff...
Compressed 1046704 bytes to 612763...
Wrote 1046704 bytes (612763 compressed) at 0x00000000 in 15.5 seconds (effective 540.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57
```

patience4711, 2023-04-13:
@bs-github  Done..

closed 2023-04-14 by bs-github

bs-github, 2023-04-18:
I can confirm that the version ESP32-ECU_v0_1a compiled for the `NodeMCU-32S` runs on that hardware as well. 

To be able to compile (I'm doing this on OS X) I had to rename the following files on disk and in the `.ino` to lower case `.h`.

- #include "HTML.H"
- #include "AAA_MENUPAGE.H"
- #include "PORTAL_HTML.H"

And obviously I had to install the missing libs.

Thanks @patience4711 !