patience4711/ESP32-read-APS-inverters #4: development

patience4711, 2023-04-19:
I finished the testing of this software and ended up wit a new version. There where lots of problems to solve, including system crashes at certain string operations like strncat and strncpy. 
Anyway, now it's working fine, [here is version 0-1b](https://github.com/patience4711/ESP32-read-APS-inverters/blob/main/ESP32-ECU_v0-1b.bin)

JanBoos, 2023-04-20:
Update naar 1b geeft bij mij een foutmelding:  [HTTP ERROR] Bad Request

[image]

patience4711, 2023-04-21:
@JanBoos Can you tell what exactly goes wrong, is this after uploading the file? (after uploading you should reboot)

I flash all my tests and experiments via ota, must be a hundred times and it always works. Could you try it again with this version?

JanBoos, 2023-04-21:
With 1c I get the same error (also with a different wifi-router). I also tried the ota with an esp8266 (9.8b) and that works ok. I read on the elegantota forum that more esp-32's have this problem. Could it be the partition-table.bin used?

Furthermore, the software on the esp32 works a lot smoother!!!

Unfortunately, my 3 APS DS3 inverters will not arrive until the beginning of May.

patience4711, 2023-04-21:
@JanBoos 
 I managed to get rid of that annoying elegantOta and installed my own updatemanager. 
For me this works fine. I hope that your problem is solved with it.<br>

[ version 1d](https://github.com/patience4711/ESP32-read-APS-inverters/blob/main/ESP32-ECU_v0-1d.bin)<br>

JanBoos, 2023-04-21:
It functions!! I first flashed 1d serially and then updated 1d again with the new OTA update. 

Thank you. Now I don't have to take my ECU + zigbee apart again and again :).

patience4711, 2023-04-22:
Yes that's the idea of OTA.  
B.t.w.  _the software on the esp32 works a lot smoother!!!_
What do you mean by that? Smoother than ?

JanBoos, 2023-04-22:
This means that the ECU software on the ESP32 responds much faster than on the ESP8266

patience4711, 2023-04-26:
The latest state of the art is [v0_2](https://github.com/patience4711/ESP32-read-APS-inverters/blob/main/ESP32-ECU_v0-2.bin)

I improved the details webpage and solved some problems with crashes during polling.

patience4711, 2023-05-09:
I tinkered a bit on the engine, resulting in a more efficient program. Moreover a re-installed the console as it is very handy if you don't have a computer nearby the ESP32-ECU

[version 0-3](https://github.com/patience4711/ESP32-read-APS-inverters/blob/main/ESP32-ECU_v0-3.bin)

If you update your existing ESP32-ECU with this version, please put off the "auto polling" (settings) first.

patience4711, 2023-07-07:
I am now testing with version [ESP32-ECU-v0_6](https://github.com/patience4711/ESP32-read-APS-inverters/blob/main/ESP32-ECU_v0-6.bin)
I introduced a new way of communication between server and webpage. Instead of frequently unnecessary requests to the server and page reloads, the server notifies the page when new data becomes available, or something has changed in the state (zigbee down, sleepmodus, etc).  The webpage only makes a request than. This reduces the wifi trafic and the cpu dramatically.  Also there is no simultaneous polling and data request handling anymore, which reduces the chances of memory erosion.

EntryGoon, 2023-10-09:
Found here that using ESP-Flasher (based on Pyflasher) is the only way to get the ESP32 flashed. unfortunately, I've spent the last hour, to found the web page credential without success... the password is "0000" by default, but the login is not defined anywhere and I can not log in the web page... And I'm pretty sure I just missed the information somewhere... ? thanks a lot for that great work !!!

YellowSub, 2023-11-03:
Bonjour,

I'm using your software in ESP-ECU-v7_01 with Wemos D1 for 3 YC600 inverters since more than one year wihout issue, but now I have 3 other inverters DS3-L;  I've upgraded to V10_8 but my DS3-L are paired but not polled (my 3 YC600 are always ok).
I' using a CC2531 . Shall I change the software of the CC2531 ( actual is CC2530ZNP-with-SBL.hex) ?  I saw somewhere a  CC2531-DS3 version. It's not possible to poll YC600 and DS3 together?

Thank you very much for sharing...

closed 2024-06-05 by patience4711