patience4711/ESP32-read-APS-inverters #5: trying to flas esp32

defaliz, 2023-04-22:
Hi,
thanks for your job ! ESP32 is a good idea !
i try to flash esp32 with this command :
/usr/bin/esptool --port /dev/ttyUSB0  write_flash 0x00000 ESP32-ECU_v0-1d.bin
i've no errors..
Wrote 992256 bytes at 0x00000000 in 98.2 seconds (80.8 kbit/s)...
Hash of data verified.

is it good ?

but i cannot see the ssid ESP-ECU, i've try with a smarphone and a laptop..
do you know why ?
regards

defaliz, 2023-04-22:
i've try flash with this tool https://github.com/Jason2866/ESP_Flasher/releases
and now that's work fine...
thanks
Alain

patience4711, 2023-04-24:
@defaliz  Likely, after the first flash it was already connected to your wifi with old credentials. Can you mention here which version you used of the ESP_Flasher? This could help others.

defaliz, 2023-04-24:
hello,
i used this soft https://github.com/Jason2866/ESP_Flasher
esp_flasher 1.4.1 it's a python program with GUI
unzip Ubuntu.zip then launch sudo ./ESP-Flasher
works fine form me..
I use rpiecu with raspberry pi since several months i've 2 ASP DS3L and 4 panels
now i try esp32 and store datas in influxdb, but i've only installed the binary on my ESP32Wroom, because my installation is 200km from me.. i could try in 2 weeks and gave you a feedback... 
thank a lot