patience4711/ESP32-read-APS-inverters #6: time not set

defaliz, 2023-05-03:
Hi,
i've try esp32 ecu near Paris and the time was good...
now i've installed esp32 ecu in Normandy and time is not set.. here is my time config 
latitude : 49.940 and longitude 1.330
and my info page
thank for help

system time = 00:12 hr.   no dst set
firmware version : ESP32-ECU_v0_2
time retrieved today : no
the wifi signalstrength (rssi) = -53
mosquitto not configured
check the mosquitto settings
system up time: 0 days 0 hrs 12 min.
current errorCode = 10

defaliz, 2023-05-03:
now the time is correct 
polling   from 08:53 hr to 17:03 hr
i've modify timezone in menu...  +120 to +60 and the time was OK after that...
about polling...
now it is 8 pm and there still sun... but polling is stopped... how can i change that ?
regards

defaliz, 2023-05-04:
hello,
just a small detail.. in page inverters output, current voltage is noticed dc voltage...
thanks

unit | panel 0 | panel 1 | panel 2 | panel 3
-- | -- | -- | -- | --
dc voltage | 35.1 | 35.1 | n/e | n/e
**dc voltage**

defaliz, 2023-05-04:
i think there is a bug...
when i reboot esp32 he lost timeconfig  (polling from 00:00 to 00:00 and run in nightmode
if i change timezone, the hour become good...

patience4711, 2023-05-05:
@defaliz It's not a bug, sometimes it happens that the time is not retrieved at once, that is always corrected in the next healthcheck (10 minutes).  I am working on the next version, the problem with dcvoltage exists in the ESP project too, nobody saw it......
I trust that for the rest it is working oke?

defaliz, 2023-05-05:
Thank for replies..
So i'll wait for the next esp32 version..
Meanwhile, I used for 6 months rpiecu with pi zero and it works very well with two inverters and for panels..

patience4711, 2023-05-06:
Well, the RPI imlementation is unbeatable. Sadly I had to dismantle it to be able to test with the ESP and ESP32 .

I still don't know if there where other issues at your side but the testversion running at a friend is working really good. More than a week error-free!

[image]

defaliz, 2023-05-06:
well, maybe a problem with my esp32.. i've bought one with a connector for an antenna..(maybe somme difficulty to retreive time ??)
i'll test soon and tell you the result..
great thanks !

patience4711, 2023-05-08:
@defaliz  when the time can't be retrieved it must be the wifi connection, maybe it is bad. If you open a serial monitor you'l see tons of debug info during the boot-up. You should be able to find out where it goes wrong i think.

closed 2023-06-17 by patience4711