patience4711/ESP32-read-APS-inverters #7: License of this project and integrating into ESPHome

derrohrbach, 2023-06-09:
Hi, first of all thank you for all the hard work you already put into this and the other APS ECU projects! It really got me started with my own project because i could validate the communication with the inverter before getting started and use your implementation as a guide!
I am currently writing my own APS ECU integration for ESPHome (https://esphome.io/). My integration is based on your code for the communication with the zigbee module and the parsing of the responses. For now my project is private but i am thinking about publishing it when it is completed (so others can use and enjoy it too). Since i could not find the license for your project i have to assume it is proprietary. 
Therefore i want to ask you: Do you allow me to publish my code on GitHub and maybe even integrate/pull it into the ESPHome project in the future, if it ever gets that far? Ideally I would like to publish my code/project under the MIT or GPL license.

Oh! And of course i would mention you and your project in my README etc. and would never claim it is all my work. Again i am very thankful for all the reverse engineering and other work you and others did.

patience4711, 2023-06-16:
@derrohrbach  I just added the MIT licence to the project so you can do what you like. 

Good luck with your project.

derrohrbach, 2023-06-16:
@patience4711 This is great news! Thanks! If you want I'll send you a link to the project.

closed 2023-06-16 by derrohrbach

derrohrbach, 2023-06-17:
In case you are interested: This is my project https://github.com/derrohrbach/esphome-apsystems