patience4711/ESP32-read-APS-inverters #8: What firmware to use? How to check if the Zigbee Module is working?

Jeroen88, 2023-06-19:
Hi @patience4711 ,

I am using a CC2530 board and an ESP32 microcontroller to read out a DS3 inverter. I was able to succesfully flash the firmware onto the CC2530 board. However, I can not retrieve data from the inverter yet. The ping sent by the firmware is not replied. 

In file cc25xxfirmware ds3 contains 3 .hex files. CC2530ZNP_2591-with-SBL.hex, CC2530ZNP_2592-with-SBL.hex and DS3_CC2530ZNP-with-SBL.hex. Which one is the correct one to use?

In the WIKI the picture of the module shows that pin P0_3 is RX and pin P0_2 is TX. Should RX be connected to the ESP32 Serial TX? Or should RX be connected to RX?

Is there a way to check the firmware running on the Zigbee module? I am thinking of sending something directly to the Serial port (preferably text) to see if it answers. Is that possible?

Do you have any other suggestion to find out why I do not get a reponse form the module?

Thank you so much!

patience4711, 2023-06-19:
@Jeroen88  if you have the cc2530 (with equal header rows) like in the image, you should flash  DS3_CC2530ZNP-with-SBL.hex. 
P03 is connected with RX on the ESP, like in the drawing.
Please read "testing and debugging" and "troubleshooting" in the wiki.
Via the console or serial monitor you can send custom commands and analyze the answers (if any).
Good luck.

Jeroen88, 2023-06-20:
Thnx @patience4711 I will try that.

closed 2023-06-20 by Jeroen88