patience4711/ESP32-read-APS-inverters #9: reboots during day

JanBoos, 2023-06-20:
I have 3 APS DS3 EU inverters, ESP32 + CC2531:
wifi: -70db
APS0: 59.6%
APS1: 49.4%
APS2: 36.9%

Everything works as it should, but I have a reboot every day (for example today at 11:56).

The main problem with that is -as far as I can see- that the cumulative total revenue also resets.

patience4711, 2023-06-20:
@JanBoos  Well the reboots is a problem but the total energy should not affected by that. That value comes form the inverter. 
You can upgrade to v0_5, should be stable.

JanBoos, 2023-06-21:
Only 1 reboot yesterday. Not clear why. 

I have the website open on 3 devices (PC, laptop and mobile). 

I am using MQTT format 2.

For the rest it works stable :)

I will upgrade to 0.5 today

JanBoos, 2023-07-02:
I still see sometimes "strange" totals. With the "old" 0.5 I went on a holyday for a week. After this week the total was 80 kWh (max per day is 10-15 kWh). ZB reset counter was 1. No further ecu reboots.

After a reboot everything was back to normal.

Put the "new" 0.5 on it yesterday. After reboot the total was only 0.12 kWh where it should be 3.8 kWh.

Does the total come from the inverter itself of by your software anyhow?

patience4711, 2023-07-03:
The total comes from the inverter or from the software, the highest value of these two. So when the ESP reboots we have the value of the inverter and when the inverter reboots (happens sometimes in twilight) the software remembered the total. 
If polling failes (inverter stopped working at sunset), every value is set to n/a. You can prevent that by putting a bigger value in poll offset. When the polling stops before the inverter goes down, there is no error situation, However, the total should be oke in all situations. In the ESP8266 this works oke.

Normally all values are reset to null at midnight so at that point there must be something wrong.

I am currently testing with th esp8266 implementation but i expect that within a week i can work on the esp32 again and correct the imperfections.

JanBoos, 2023-07-03:
I mean totals per inverter.

The total of all inverters is (also wrong, but) indeed the sum of the inverters.

JanBoos, 2023-07-08:
Thanks for 0.6. That indeed works better that you no longer have to refresh the browser.

Incidentally, after taking the PC out of sleep mode, sometimes a strange total (see photo). Refreshing the browser (google chrome) a few times and the values ​​are correct again.

Furthermore, since 0.5a, no strange resets have been seen from both the ecu and the zigbee.

Also nice (for the administration) that the values ​​are saved after the polling has stopped.

[image]

patience4711, 2023-07-18:
@JanBoos  I trust that this "impossible" phenomenon has gone in the newest version.

closed 2023-07-25 by patience4711