patience4711/ESP32-read-APS-inverters #12: the zigbee module is not working - How to fix ?

ds2k5, 2023-11-23:
Hello,

[image]

flashed the file: CC2530ZNP-with-SBL.hex to the CC2530

Wiring:

**ESP32      CC2530**
  3.3V    <->  3.3V
  GND    <-> GND
  GPIO5 <-> RST
  GPIO7 <-> R03
  GPIO8 <-> R02

ESP32

https://www.flickr.com/photos/jgustavoam/40089095211

CC2530

https://www.zigbee2mqtt.io/assets/img/webee_cc2530_cc2591_pinlayout.c476b9b5.png

What did I wrong ?

patience4711, 2023-11-23:
The wiring is wrong, the rx2 and tx2 pins are gpio16 and 17 please read the wiki (hardware).  If it still doesn't work please look in the isues for the esp-ecu [like here](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/issues/98) this is a problem that keeps popping up, its always the wiring or improper firmware.

ds2k5, 2023-11-24:
thanks.... looks like I killed my CC2530 :-(

patience4711, 2023-11-24:
Thats bad news, I thought you had it flashed already. You can only fry it when vvc too high.

ds2k5, 2023-11-24:
yes I flashed with a SmartRF

I reversed 3V and GND by connecting to the ESP32
looks like the CC2530 did not have a reverse polarity protection

patience4711, 2023-11-24:
Too bad, i hope you have better luck next time.

ds2k5, 2023-12-12:
got a new cc2530 flashed the firmware file: CC2530ZNP-with-SBL.hex but had the same issue: the zigbee module is not working.

[image]

[image]

https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/cc25xx_firmware.zip

ESP32 CC2530
3V <-> 3V
GND <-> GND
D5 <-> RST
RX2/GPIO16 <-> P03
TX2/GPIO17 <-> P02

[esp32-ecu-0.8.zip](https://github.com/patience4711/ESP32-read-APS-inverters/files/13650055/esp32-ecu-0.8.zip)

patience4711, 2023-12-15:
Before anything else you should ensure that the zigbee is working, (console health) 

I don't know what you did here, looks like you have registered 8 inverters. To get rid of them you can visit the console and click "help". 10;erase removes all inverterfiles.

ds2k5, 2023-12-15:
>Before anything else you should ensure that the zigbee is working, (console health)

looks good did not give the message again

>I don't know what you did here, looks like you have registered 8 inverters. To get rid of them you can visit the console and click "help". 10;erase removes all inverterfiles.

ENTER S/N of Inverter and klick to add button

patience4711, 2023-12-19:
I would like to close this issue with a conclusion that is helpfull to others. The problem was that the zigbeemodule didn't work. So what is the present state, does it work now? What was the solution?

ds2k5, 2023-12-19:
>I would like to close this issue

ok thanks for your help

> What was the solution?

may I did not wait long enough

closed 2023-12-22 by patience4711