patience4711/ESP32-read-APS-inverters #17: how to flash this firmware

jmguilbert, 2024-01-10:
Good morning
Could any of you help me on how to flash an ESP32 with ECU firmware?
With Ubuntu or Linux.
I've been trying for hours but haven't succeeded.
I succeeded in my tests with an 8266 and even a Pi Zero. But impossible with the ESP32.
Help me please.
Jean-Marc.

jmguilbert, 2024-01-11:
I'm answering this myself because I finally found a reproducible procedure.
From an ESP32, whatever it may be:
1- use the Arduino IDE, choose the right card and flash with the example (in “files”) OtaWebUpdater.
2- find the IP address of your ESP32 in your router.
3- connect to the ESP32 web page (admin/admin) and flash it with the file: ESP32-ECU_v0-8.bin
4- in principle, the ESP32 will take a new IP and it will be the ESP-ECU. 
Login: admin
Password: 0000

I hope the time spent on this will be of use to someone else. I am now waiting for the sun to install my solar panels and activate the service on my APsystem DS3
Jean-Marc. (in France)

Hansiart2, 2024-01-14:
@jmguilbert  Bonjour Jean-Marc thanks for this contribution. This is a very smart approach.

It seems that more people have troubles to flash the ESP32 so i think is surely would help others.

closed 2024-06-05 by patience4711