patience4711/ESP32-read-APS-inverters #19: CONNECT TO WIFI VIA BLUETOOTH

patience4711, 2024-02-06:
To simplify the wifi connection procedure, i am experimenting with the bluetoothserial library. As an alternative for the complex and sometimes unreliable captive portal.
This works very good: When the wifi connection fails, the bluetooth is started (the blue led goes on) and we can connect to the ESP32 via an app on our phone. Here we can set the wificredentials, the security level and the admin password.

The enormous overhead of that wifimanager can be cleaned up this way.

APSjens, 2024-05-04:
Hello patience4711,

First I want to say thank you very much for developping such an excellent program...
I just tried the new bluetooth version but had not luck with the authentication. After "authentication" the esp32 answers with "please enter your admin password". Wether I put in the chipid or the long number, it is accepted! The answer is: "You entered 123..." But in the same moment the esp32 says "wrong passwd". What did I wrong? Best regards, Jens

APsystems YC600, ESP32-WROOM-32U, CHD tech CC2530,

APSjens, 2024-05-16:
Hi patience4711,
to be more precise it is that I'm not able to put in the right password. It is not the chipid or 12346789. Could you give me some hints what is going wrong? Best regards, Jens

patience4711, 2024-05-29:
Did you check that the cr and lf are disabled in the serial-bt app ?

APSjens, 2024-05-29:
Thanks a lot, that was it!

closed 2024-06-05 by patience4711