patience4711/ESP32-read-APS-inverters #20: ESP32 + CC2530 with DS3L - pairing failed

ds2k5, 2024-05-11:
Hello,
the pairing failed.

FW File: ESP-ECU-v10_8.bin

flashed the file: CC2530ZNP-with-SBL.hex to the CC2530

Wiring:

ESP32 CC2530
3V <-> 3V
GND <-> GND
D5 <-> RST
RX2/GPIO16 <-> P03
TX2/GPIO17 <-> P02

Steps I did:

0. conntect the ESP32 to the local WiFi
1. Menu -> Inverter -> add -> enter Serial & save
2. click to pair

there is a part of the console output:

```
  sendZB FE00260026
readZB FE0166050062FE064180020202020702C2FE0166050062FE0166050062FE0166050062FE0166050062FE0164000065FE00660066FE0145C0088CFE0145C0088CFE0145C0088C  rc=23

checkZigbeeRadio
  sendZB FE00270027
readZB FE0E670000FFFF80971B01A3D8000007090011  rc=19

found 0709, running oke
ZB coordinator started
trying pair inv 0
pair command 0 = 24020FFFFFFFFFFFFFFFFF14FFFF140D0200000F1100703000082926FFFF10FFFF80971B01A3D8
  sendZB FE2524020FFFFFFFFFFFFFFFFF14FFFF140D0200000F1100703000082926FFFF10FFFF80971B01A3D83C
readZB FE0164020067  rc=6

pair command 1 = 24020FFFFFFFFFFFFFFFFF14FFFF140C0201000F0600703000082926
  sendZB FE1A24020FFFFFFFFFFFFFFFFF14FFFF140C0201000F060070300008292672
readZB FE0164020067  rc=6

decoding : FE0164020067
no valid pairing code, returning...
pair command 2 = 24020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100703000082926A3D810FFFF80971B01A3D8
  sendZB FE2524020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100703000082926A3D810FFFF80971B01A3D844
readZB FE0164020067  rc=6

decoding : FE0164020067
no valid pairing code, returning...
pair command 3 = 24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D8
  sendZB FE1A24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D84F
readZB FE0164020067FE0164020067  rc=6

failed, inverter got id 0000
Opened for write..../Inv_Prop0.str
send normal ops initCmd = 2401FFFF1414060001000F1E80971B01A3D8FBFB1100000D
  sendZB FE162401FFFF1414060001000F1E80971B01A3D8FBFB1100000D4F
readZB FE0164010064FE034480001401D2  rc=14

zb initializing ready, now check running
checkZigbeeRadio
  sendZB FE00270027
readZB FE0E670000FFFF80971B01A3D8000007090011  rc=19

found 0709, running oke
300 secs passed, polling300000

```

[image]

[image]

[image]

[image]

[image]

[image]

[image]

APSjens, 2024-05-19:
Hi ds2k5,
I had the same issue and for me the following worked:
In console put in: 10;ERASE
That deletes the inverter file.
After this it was possible to register the inverter. Before I had to do put in the data again and again and it did not save this.
Then you can pair. What also doesn't work is the automatic polling. In the moment I'm doing this by 10;poll=0. No other chance.
Good luck, Jens

patience4711, 2024-05-22:
@ds2k5 Hard to say why the pairing fails, the coodinator works oke but you are not getting the right answers from your inverter. 
@APSjens when the autopoll failes, ther must be something wring in the settings that are stored in the filesystem. When you boot and examen the serial monitor you can see the value of 'polling' in basisconfig.txt. If this value is 1 the autopoll will work.

ds2k5, 2024-05-22:
thanks @ALL

but I can not test before End of August...
then I will be there next time...

ds2k5, 2024-08-25:
@APSjens 

Hi,
I am back at the Place.

Tyed what you told me.

  
```
* * connection opened * *
--
poll inverter 0
pollCommand ex checksum:240100001414060001000F1380971B01A3D8FBFB06BB000000000000C1FEFE
sendZB FE1D240100001414060001000F1380971B01A3D8FBFB06BB000000000000C1FEFE29
readZB   FE0164010064FE034480001401D2FE274481000006000000141400FF0000000000001380971B01A3D8FBFB06BB000000000000C1FEFE00000002   rc=58
decodePollAnswer inverter 0
no route receipt
ignoring,  received  FE0164010064FE034480001401D2FE274481000006000000141400FF0000000000001380971B01A3D8FBFB06BB000000000000C1FEFE00000002
polling failed with errorcode 15

```

ds2k5, 2024-08-26:
Something changed:

[image]

[image]

[image]

But did not see Data how much kw/h are done by the inverter
or how much power current is made...

ds2k5, 2024-08-26:
Thanks to all.... move the ESP32 to another place
now the data is show

[image]

ds2k5, 2024-08-26:
shit..... it stopped working :-(

[image]

ds2k5, 2024-08-27:
What did Inverter Status: 1111 mean ?

[image]

Did delete and reAdd the inverter
but pairing did not show success....

how to factory reset the device ?

closed 2024-11-12 by patience4711