patience4711/ESP32-read-APS-inverters #21: Inverter Page unusable after adding the 9th inverter

jeffajst, 2024-05-15:
Hi,

thanks you for this great project and all your hard work.

I have a total of 9 inverters (6x DS3 and 3x QS1) connected. After adding the 9th inverter /INV_CONFIG page becomes unusable.

It looks like this:
[image]

Nothing is clickable anymore, inverter data not editable anymore. Everything is fine until the 9th inverter is added.

I managed to pair it by manually navigating to:
- /INV?welke=8
- /PAIR

I am suspecting a bug here, but not really sure. Maybe I did something wrong.

Is there any reason for the limit of a maximum of 9 inverters? I have a friend with about 16-20 inverters in use.

patience4711, 2024-05-22:
Yes there is  a limit of 9 inverters, this is not so easy to extend. With an installation like that it maight be preferable to buy a real ECU. Or multiple ESP ecu's.

jeffajst, 2024-05-22:
I do have a real ECU, but I don't want to use APSystems Cloud. I have a total of 9 inverters (6x DS3 and 3x QS1) connected. I used [ESP32-ECU_v0-8](https://github.com/patience4711/ESP32-read-APS-inverters/blob/main/ESP32-ECU_v0-8.bin) and a CC2530 with "CC25xxfirmware ds3/CC2530ZNP-with-SBL.hex" from [cc25xx_firmware.zip](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/cc25xx_firmware.zip).

Pairing and Polling works, the only issue I have so far is that the /INV_CONFIG page becomes unusable after adding the 9th inverter as described in my first comment here.

patience4711, 2024-05-25:
Oke, next week i have time to investigate this.

closed 2024-06-05 by patience4711