patience4711/ESP32-read-APS-inverters #23: How can I read out the daily production?

ds2k5, 2024-08-26:
Hi,
thanks for you great Software.
Got it work today with a apsystems DS3-L.

Like to build a script that grap daily the daily production power
to bulid a overview.
But I have trouble to read the source code of the main page
```

<!DOCTYPE html><html><head><meta charset='utf-8' name='viewport' content='width=device-width, initial-scale=1'><title>ESP32-ECU</title><link rel="icon" type="image/x-icon" href="[/favicon.ico](view-source:http://192.168.144.14/favicon.ico)" /><link rel='stylesheet' type='text/css' href='[/STYLESHEET](view-source:http://192.168.144.14/STYLESHEET)'><style>body { background-color: #EEE;}span {padding: 6px;}table, th, td {border: 2px solid blue; font-size:16px; padding:6px; text-align:center; border-collapse:collapse;backgound-color:#dfff80;}tr {background-color:#ccffcc;}td { width:70px; }.btn { background-color: #199319; color: white; padding: 5px 22px; border-radius:6px;}.btn:hover {background: #eeeF; color:black;} @media only screen and (max-width: 800px) {th, td { width:60px; font-size:11px;}tr {height:35px;} .btn { padding: 5px 18px; font-size:10px;}}</style>
<script type="text/javascript" src="[SECURITY](view-source:http://192.168.144.14/SECURITY)"></script>
<script type='text/javascript'> 
var term,table_row;window.addEventListener("visibilitychange",(()=>{"visible"===document.visibilityState&&(getGeneral(),setTimeout(getAll,300))}));var cnt=0,totalEn=0;function loadScript(){getGeneral(),setTimeout(getAll,300)}function celbgc(e){e.startsWith("e")||e.startsWith("i")?document.getElementById(e).style="background-color:#c6ff1a":document.getElementById(e).style="background-color:#a6a6a6"}function getAll(){totalEn=0;for(let e=0;e<cnt;e++)term="get.Power?inv="+e,table_row="inv"+e,getData(e)}function getData(e){var t=new XMLHttpRequest;t.onreadystatechange=function(){if(4==this.readyState&&200==this.status){var t=this.responseText,n=JSON.parse(t),l=n.eN,o="r"+String(e);document.getElementById(o).style.display="table-row";var a="i"+String(e);document.getElementById(a).innerHTML="<a href='/details?inv="+String(e)+"'><button class='btn'>"+String(e)+"</button</a>";for(let t=0;t<4;t++)a="p"+String(e)+t,"n/e"==n.p[t]?celbgc(a):"n/a"==n.p[t]?document.getElementById(a).innerHTML=n.p[t]:document.getElementById(a).innerHTML=n.p[t].toFixed(1);celbgc(a="e"+String(e)),"n/a"!=l?(cnt>1&&(totalEn+=parseFloat(l),celbgc("e9"),document.getElementById("r9").style.display="table-row",document.getElementById("e9").innerHTML=totalEn.toFixed(2)),document.getElementById(a).innerHTML=l.toFixed(2)):document.getElementById(a).innerHTML=l,document.getElementById("noOutput").style.display="none",document.getElementById("4channel").style.display="block";const d=(new Date).toString();document.getElementById("time").innerHTML=d.substr(16,8)}},t.open("GET",term,!0),t.send()}function getGeneral(){var e=new XMLHttpRequest;e.onreadystatechange=function(){if(4==this.readyState&&200==this.status){var e=this.responseText,t=JSON.parse(e);cnt=t.cnt,console.log("getGeneral cnt = "+cnt);var l=t.st,o=t.sl;0==t.rm&&(document.getElementById("ml").style.display="block"),"11"==l?document.getElementById("busy").style.display="block":(document.getElementById("busy").style.display="none",document.getElementById("failed").style.display="none"),"0"!=l&&"25"!=l||(document.getElementById("failed").style.display="block");var a=document.getElementById("maindiv");document.getElementById("sleep");"1"==o?(a.style.background="grey",document.getElementById("noOutput").style.display="none"):a.style.background="linear-gradient(#e8edc8, #c8eaed)"}},e.open("GET","get.General",!0),e.send()}if(window.EventSource){var source=new EventSource("/events");source.addEventListener("open",(function(e){console.log("Events Connected")}),!1),source.addEventListener("error",(function(e){e.target.readyState!=EventSource.OPEN&&console.log("Events Disconnected")}),!1),source.addEventListener("message",(function(e){console.log("message",e.data),"general"==e.data&&getGeneral(),"getall"==e.data&&getAll(),"reload"==e.data&&(getGeneral(),setTimeout(getAll,300))}),!1)}
</script></head><body onload='loadScript()'>
<div id='msect'><div id='menu'><a class='fright' id='ml' href='[/MENU](view-source:http://192.168.144.14/MENU)'>menu</a></div>
<kop>ESP32 ECU</kop><div class='divstijl' id='maindiv'><center><p>

<h4>POWER / ENERGY @ <span id='time'></span></h4><div id='noOutput' style='display:block'><h4 style='color:red'>waiting for output</h4></div>
<div id='4channel' style='display:none;'><center><table><tr style='Background-color:lightblue; font-weight:bold; text-align:center; border:4px solid black;'>
<td>inverter<td>panel 0<td>panel 1<td>panel 2<td>panel 3<td>kWh</td></tr>
<tr id='r0' style='display:none;'><td id='i0'></td><td id='p00'></td><td id='p01'></td><td id='p02'></td><td id='p03'></td><td id='e0'></td></tr>
<tr id='r1' style='display:none;'><td id='i1'></td><td id='p10'></td><td id='p11'></td><td id='p12'></td><td id='p13'></td><td id='e1'></td></tr>
<tr id='r2' style='display:none;'><td id='i2'></td><td id='p20'></td><td id='p21'></td><td id='p22'></td><td id='p23'></td><td id='e2'></td></tr>
<tr id='r3' style='display:none;'><td id='i3'></td><td id='p30'></td><td id='p31'></td><td id='p32'></td><td id='p33'></td><td id='e3'></td></tr>
<tr id='r4' style='display:none;'><td id='i4'></td><td id='p40'></td><td id='p41'></td><td id='p42'></td><td id='p43'></td><td id='e4'></td></tr>
<tr id='r5' style='display:none;'><td id='i5'></td><td id='p50'></td><td id='p51'></td><td id='p52'></td><td id='p53'></td><td id='e5'></td></tr>
<tr id='r6' style='display:none;'><td id='i6'></td><td id='p60'></td><td id='p61'></td><td id='p62'></td><td id='p63'></td><td id='e6'></td></tr>
<tr id='r7' style='display:none;'><td id='i7'></td><td id='p70'></td><td id='p71'></td><td id='p72'></td><td id='p73'></td><td id='e7'></td></tr>
<tr id='r8' style='display:none;'><td id='i8'></td><td id='p80'></td><td id='p81'></td><td id='p82'></td><td id='p83'></td><td id='e8'></td></tr>
<tr id='r9' style='display:none;'><td colspan='5' style='text-align:right;'>total of all inverters&nbsp;</td><td id='e9'></td></tr>
</table></div><p>Powered by Hansiart</p><div id='busy'><span style='color:red;'><h3>checking / initialyzing zigbee network...</h3></span><br></div><div id='failed' style='display:none;'><span style='color:red;'><h3>the zigbee module is not working.</h3></span><br></div></center></div></div></body></html>
```

[image]

Any hint how to grap the "0.44" in this example ?

Is it possible to see how much the inverter has produced since connected to to solar panals ?

patience4711, 2024-09-04:
When you issue a link like http://ip_of_esp_ecu/get.Inverter?inv=0 you'l get a json with all data, you can grab any value from this.

The inverter is reset when it is dark so no values are retained. The ESP32 doesn't remember anything because that would be gone at every reset.

ds2k5, 2024-09-04:
thanks - I will be back frist middle next year

I will try then