patience4711/ESP32-read-APS-inverters #24: The zigbee module is not working...

jeremypsl, 2024-09-29:
Hello,

I try to make working a CC2530 with an ESP32 but i'm stuck with "the zigbee module is not working."
I flashed the CC2530 with a CCDEBUGGER and SmartRF Flash Programmer v1.
I wired VCC to 3V3, GND to GND, P02 to TX2 (GPIO17), P03 (GPIO16) to RX2 RST to D5 (GPIO5).

I don't know how i can be sure my flash worked on the CC2530.

I don't know how i can be sure my wiring is ok.

Please help !

patience4711, 2024-10-11:
How about some more details:
What esp32 do you use?
What cc2530 model and which software did you flash on it?

If you read the wiki you find the answer about the wiring in the chapter "hardware".
If you read the wiki you find the answer about the testing of the ZB module in "troubleshooting".

Furthermore you can look in the issues if the esp8266 project [here](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/issues)

jeremypsl, 2024-10-13:
I use : 
- HiLetgo ESP-WROOM-32 ESP32 ESP-32D Development Board
- Wisboey ZigBee Module de PA TTL Uart Port SéRie de Conversion CC2530 + CC2591 Module Zigbee CC2530 Dev Board

I flashed the CC2530ZNP-with-SBL.hex, CC2530ZNP_2591-with-SBL.hex then CC2530ZNP_2592-with-SBL.hex.

I followed the wiki for wiring and try others things but none worked...

patience4711, 2024-10-14:
On the pinout of that board i can't find RX1 and TX1 so maybe first determine whether this uart is really available. The wiki tells you how can do the uart loopback test. If this doesn't work you should buy another esp module (make sure it has RX1 and TX1).

If the hardwaretest works you should flash the  CC2530ZNP-with-SBL.hex on the zigbee and try the software loopback test.
If that fails, your zigbee must be defect or one of the dupontcables is interrupted. You can test your cables by using them one by one for the hardware loopback test.

good luck.