patience4711/ESP32-read-APS-inverters #25: The zigbee module is not working.

Yourry, 2025-01-15:
Hello,
I am very interested in this project of reading DS3 inverters, I have a gatewau ECU-B which does not work anymore..
I normally did all the requested installations and I arrive on the principle page with the following error:

> the zigbee module is not working.

I read carefully the advice in the following link which presents the same problem:
https://github.com/patience4711/ESP32-read-APS-inverters/issues/24
When I launch the following command:

`10;zbt=2710AABBCC`

I have the following response:

> * * connection opened * *
going to send a teststring, len=17
the command = 2710AABBCC
sendZB FE032710AABBCCE9
readZB FE036710AABBCCA9 rc=8 

So I'm already wondering if the zigbee module is working properly?

Because the ECU-B is no longer working, I couldn't delete the pairing, maybe it's my problem but I don't understand why so I have this message on the main page : the zigbee module is not working.

If you have an idea or test that I can do, I'm interested.
Thank you very much for your help.

Yourry, 2025-01-27:
Great, thanks for your answer.
I'm tearing my hair out over this problem because it annoys me not to have an ECU-B anymore.
Regarding my problem, if I run the command from the console:

> * * connection opened * *
command = 10;INIT_N
starting coordinator
init zb coordinator
ZB module hard reset
cmd : 0
zb send cmd 0
sendZB FE03260503010321
readZB FE0166050062 rc=6
cmd : 1
zb send cmd 1
sendZB FE0141000040
readZB FE0166050062FE064180020202020702C2 rc=11
cmd : 2
zb send cmd 2
* * connection closed * *

The logs via the serial port are perhaps more talkative when starting the module:

[log_start.txt](https://github.com/user-attachments/files/18563852/log_start.txt)

For the zigbee module, it is this model:

https://ae01.alicdn.com/kf/S009c5dcea1f74c02a3193b0a519c2ed76.jpg

It is flashed with the firmware:

`\read-APSystems-YC600-QS1-DS3-main\cc25xx_firmware\CC25xxfirmware ds3\CC2530ZNP-with-SBL.hex`

-> CC2530ZNP-with-SBL.hex

The inverters are DS3-L

The firmware flash was done with TI's CC-DEBUGGER and its program. It tells me it's OK

Finally I even tested a variant with the cc2530 + cc2591 and the firmware:
CC2530ZNP_2591-with-SBL.hex
Still doesn't work.

Thanks a lot

patience4711, 2025-01-28:
The configuration you had when you recorded the log_start.txt is the right one because the zigbee reacts adequate on the start commands.  That seems completely fine. 

In my system the answer to the 270027 command is FE0E670000FFFF80971B01B3D600000709000F that looks like your answer exept for the 80971B01B3D6 part. That is the id of your coordinator , you seem to have a coordinator with id 000000000000 which is invalid.

Now in your log,  cmd 2 is missing the reversed ecu-id (should be FE0A26050108FFFF followed by 80971B01B3D6)  and yours is FE0A26050108FFFF. I think that in the settings the ecu-id is missing. This ecu-id field has a default value so you must have wiped it somehow.
It should be something like D6B3011B9780 12 characters. I think it is wise to recheck all the settings.

Yourry, 2025-01-30:
I'm already happy to see that it's not a problem with the zigbee.
Because by setting up a Zigbee sniffer I had no frame but as I don't master this radio protocol at all, I must have made a mistake.
In the meantime, here is my inverter config page and the ID is correctly entered:

[image]

Here is the associated log when I start the esp32 :

[log_start.txt](https://github.com/user-attachments/files/18607858/log_start.txt)

It's strange to see this line:

> {"ECU_ID":"ÿÿÿÿÿÿÿÿÿÿÿÿÿ%","userPwd":"1111","inverterCount":1,"Polling":true,"pollOffset":0}

No?

patience4711, 2025-01-31:
You are confusing inverter id and ecu-id go to >menu >settings

Yourry, 2025-02-01:
Yes I understood but where do I find this ECU ID?

patience4711, 2025-02-03:
If you click menu and settings in the webinterface like i said above, the you wil enter this page (last time i tried it worked).
Normally the field contains a default value as shown in the picture but somehow you managed to make it go away.

[image]

Yourry, 2025-02-05:
I have no value there :

[image]

However, I was able to add the value given as an example and it now works but it remains constantly empty.
The main thing is that it works, I already have 1 inverter out of 2 :

[image]

Thank you for your patience and the work.
Kind regards,

closed 2025-02-05 by patience4711