patience4711/ESP32-read-APS-inverters #26: Ethernet connection for esp32?

dim6003, 2025-04-15:
Hi, nice project, willing to give it a try very soon, as I just bought a ds3-h inverter, but ecu seems quite expensive for what it does.
However, I'm not fond of using WiFi for home automation, so I would like to use an Ethernet enabled esp32 module, such as:
[](https://a.aliexpress.com/_EJ1kZ68)
Or
[](https://a.aliexpress.com/_EGJsaLw)
However I was not able to find specific data about expected esp32 characteristics (ram, rom, gpio, ...) for this project,  so I cannot say if above links would work (on wifi as first step, and in Ethernet after sw adaptation)?
Or would it be easier to go for a more 'standard' esp32 module, and connect it to some w5500 Ethernet adapter [](https://a.aliexpress.com/_Eu6gnfm) or similar?
Thanks!

dim6003, 2025-04-18:
Or would it be easier with the pico-pi branch, tas here are some pico-pi with integrated Ethernet...?

patience4711, 2025-04-18:
What you want involves a lot of programming and testing. Pico uses micropython which is totally different from the C that is used for the other platforms. Unless you can program yourself (or someone else, not me) this is a dead end.
I don't see the problem with wifi but you must have your reasons i guess.
The simplest solution would be an old raspberry B with an ethernet Port. This has a lot more features than the ESP. [look here](https://github.com/patience4711/RPI-APS-inverters)
Maybe you can buy a second hand one cheap.

patience4711, 2025-04-18:
To be honest, i am not very interested in learning micropython, too old for that. If i remember well, there is already python software that can read some APS inverters.  Thought i saw it [here](https://github.com/Koenkk/zigbee2mqtt/issues). 
Good luck.

closed 2025-05-16 by patience4711