patience4711/ESP32-read-APS-inverters #28: Cannot Compile Source V1.1

seby90, 2025-07-10:
> thk's for the updated !
> 
> Despite having installed the latest Arduino IDE and selected the correct ESP32 board package, I’m still getting compilation errors. Could you please help?
> 
> In line 113 of ESP32_ECU_v1_1.ino, the declaration (( char static_ip[16] = ""; )) has been commented out, which triggers a compilation error.
> 
> In line 141 of HTML.h, the declaration (( const char CONFIRM_IP[] PROGMEM = R"=====(  )) is commented out, causing another compilation error.
> 
> There’s a redundant double definition of DETAILSPAGE in both DETAILSPAGE.h and DETAILSPAGE.ino...  Would it make sense to simply delete DETAILSPAGE.ino altogether, since it’s over two years old?
> 
> Finally, I’m still running into compilation errors due to library versions. ESPAsyncWebSrv by darrel v 1.2.9 it's ok ?? Because if i use this lib, we have error compilation C:/Users/...../AppData/Local/Arduino15/packages/esp32/tools/esp-x32/2411/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:\Users\....\AppData\Local\arduino\sketches\1478D46A45E37B5262485B19AA829C9E\libraries\ESPAsyncWebSrv\AsyncWebSocket.cpp.o:(.literal._ZN17AsyncWebLockGuardC5ERK12AsyncWebLock[_ZN17AsyncWebLockGuardC5ERK12AsyncWebLock]+0x0): undefined reference to `pxCurrentTCB'
> 
> Could you clarify exactly which libraries (and versions) this project requires?
> 
>  

 _Originally posted by @seby90 in [#27](https://github.com/patience4711/ESP32-read-APS-inverters/issues/27#issuecomment-3059279123)_

patience4711, 2025-07-15:
some testresults computed by the software, they do not exactly match our valueas but that is a round matter of the 16.59 multiplication i think

100W  2401 0000 1414060000050F1480971B D8A3011B9780 FBFB06AA270000 067B 01 58 FEFE

300W  2401 0000 1414060000050F1480971B D8A3011B9780 FBFB06AA270000 1371 01 5B FEFE

500W 2401 0000 1414060000050F1480971B D8A3011B9780 FBFB06AA270000 2067 01 5E FEFE

I have a .bin ready to test with [here](https://github.com/patience4711/ESP32-read-APS-inverters/blob/main/ESP32_ECU_v1_2.ino.bin)
In the frontpage click the button and in the detailspage you can set the throttle value.

seby90, 2025-07-15:
Let’s answer these questions.

 First of all, in your example frame the ECU’s MAC address isn’t reversed. Please refer to my example where I used: 1414060000050F14 80971BXXXXXX FBFB06AA

So your 600 W frame becomes:
10;ZBT=2401 103a 1414060000050F14 80971B01A3D8 FBFB06AA270000 26E2  01  DF  FEFE (of course, remove all the spaces!). 

Yes—when you send the prefix 10;ZBT= from your ESP32 firmware, the ESP32 will automatically append the FE byte, the payload length, and the checksum at the end of the line!

Regarding the other clarifications you asked for:

Throttling is set indefinitely. Once you send the power-limit command, the inverter retains that setting until you explicitly change it.

ECU-C integration: If you’re using an ECU-C, the throttling information is forwarded to APS’s EMEA platform, which confirms that the command frame works correctly.

Zero-Export mode: Activation of zero-export is only available on an ECU-C. Equipped with current-measurement CTs, the ECU-C calculates and enforces the micro-inverters’ output limit based on the zero-export target you configure in the ECU.

For the fourth point, it’s more obscure and only APS can really clarify. From my experience as an end user, I once enabled zero-export and, after disabling it, my inverters never returned to full output. I had to call APS, and they told me this was expected and that you must redefine each micro-inverter’s max power afterward. What I’ve gathered is that while zero-export is active, the ECU sends regular power-limiting frames (based on CT readings) according to the target you set (e.g. 100 W max export), and when you stop zero-export, the last frame sent is saved on the inverters. To me, this seems like a bug on their side—it would be so simple to send one final frame to restore full production.

Sources

Demander à ChatGPT

seby90, 2025-07-15:
> I am interested in an eventual answer from the inverter to this command.

Yes, there is a response—I’ll capture the replies from my units and share them.”

patience4711, 2025-07-15:
So it was reversed like i suspected, i corrected that and a new binary can be found [here](https://github.com/patience4711/ESP32-read-APS-inverters/blob/main/ESP32_ECU_v1_2.ino.bin)

test 300W: 2401 0000 1414060000050F1480971B 80971B01A3D8 FBFB06AA270000 1371 01 5B FEFE

It is still not clear to me how we set the inverter back to full output. By sending a throttle value of 1000 or so?

patience4711, 2025-07-15:
Well so far i thank you for this very usefull contribution, excelent work.  Sadly i have only yc600 inverter and to my knowledge i can't throttle that. So i have to leave the testing to others.

But this is all interesting stuff, it is possible to program the ESP32ECU so that it reacts to an api to throttle all inverters say 35W down to create a zero production situation.

seby90, 2025-07-16:
All inverter models that can be paired with an ECU-C can be throttled. I’ve also owned YC600 units, and I can tell you they weren’t running at full power when I received them! Try sending the command to set them to 1000 W—you’ll see the difference.

As for your two remaining questions:

Shut-off command: There must be a command to turn an inverter off, since that option exists in the ECU-C’s hidden menus. I don’t know the exact implementation, though—the only way to find out is to send the command and sniff the Zigbee traffic to reverse-engineer it.

Risk of erratic behavior: I don’t see why the inverter would “go crazy.” It’s simply enforcing a production limit. If you set a limit above what the inverter can actually produce, it will act as a hard cap—it won’t exceed that value, but it won’t behave unpredictably.

mi-hol, 2025-07-16:
> Sadly i have only yc600 inverter and to my knowledge i can't throttle that

@patience4711 would you be interested to receive a DS3-S (restricted to 600W but it was capable to supply 849W until it got limited by EMA platform (as mentioned by seby90)) as donation ?

patience4711, 2025-07-16:
@mi-hol sure I never had a donation so i would be honoured. How do you want  to do this?

@seby90 I am implementing the throttle command in my rpi-zero implementation, when i have it ready i can experiment. I'l let you know..

seby90, 2025-07-16:
“OK, I’ve flashed the latest v1.2. FYI, the command 10;ZBT=2401xxxx1414060000050F1480971BXXXXXXFBFB06AA27000040CE01E5FEFE
is only sent over the Arduino’s serial interface. The command is truncated in the ESP32’s web console! You must be connected via USB and have console mode enabled in Arduino for sending the commande 10;ZBT=

send a zigbee command, len=69
the message = 2401xxxx1414060000050f1480971BxxxxxxFBFB06AA270000067B0158FEFE
sendZB FE1D2401xxxx1414060000050f1480971BxxxxxxFBFB06AA270000067B0158FEFEED
readZB FE0164010064FE034..........  rc=66    So the response is “RC66” when it’s validated.

patience4711, 2025-07-16:
@seby90 there is still a thing that puzzles me. In the inverterpage there is a value for the status, when paired it shows a value let's say "0xAABB" 
In the throttle command, is that '2410 AABB etc'  or '2410 BBAA etc' ( so reversed). I've send my first throttlecommand but i get FE0164010064FE034480001400D3 which are in fact 2 messages, not indicating success if i remember well.

EricMicPro, 2025-07-17:
@patience4711, I think there are still some errors in your last trial frame.
- 0x0000 for the Dst short Addr. It must be the reversed short Address of the Inverter.
   So if you have 0x14A2, you must put A214
- Maybe 0x00 for TransID is OK but better to put 0x01
- The Data Length must be 0x13 instead of 0x14
- FCS must be calculated (XOR of all bytes)

See below the frame explanation.
[image]

seby90, 2025-07-17:
@patience4711 

I flashed the latest version and was able to run some tests. To begin with—and to answer your question—when you pair a micro-inverter and its status shows, for example, B901, then your command should be 10;ZBT=2401B901… 

I also noticed that even if the MAC address configured in the ESP32 differs from the one sent in the frame, it still works!

However, when I tested the built-in “throttle” command, it doesn’t work every time—I’m not sure why. Strangely, when I confirm a power setting, I only see this appear: sendZB FE1D2401xxxx1414060000050F1480971B01A3D8FBFB06AA270000067B0158FEFEC3 found Inverter

There must be a bug. In any case, if I paste my frame directly into the console 10;ZBT=… it works flawlessly now.

patience4711, 2025-07-17:
@seby90 Oke, So the inverter shortID is not reversed.  So when you send a frame successfully, you should get an answer you can see in the console. This answer is analized by the software, the "found inverter" is a result of that but i must check that. What do you see in serial monitor? Its a real pain that i can't test myself.

@EricMicPro Good analysis.  First of all, can you confirm that you are able to send a successfull throttle command? If so, would you mind to explain a bit more. I'd like to know:
- what exactly is your inverter id looking at the inverter page 0x....
- you say that this should be reversed in the frame, (that is not the same as in a poll command and is not what seby90 says) 

The built-in calculation of length and CRC has proven itself enough. 
300 W test sent via raspberry 2401 3A10 14 14 06 00 01 05 0F 13 80 97 1B 01 B3 D6 FB FB 06 AA 27 00 00 13 71 01 5B FEFE 
exactly like yours only the ECU-ID differs  Checked the rest 
 FE1D2401 3A....5BFEFED7 and it has a different checksum, due to the deviation in ECU-ID.  Anyway it doesn't work.

patience4711, 2025-07-17:
I made a new compile, this one tries to analyze the response on every frame sent by console or serial monitor or the built-in throttle command. It shows what we get in and if it's usefull. Still we don't know what to expect on a throttle command but this will help.

As to answers, it seems that the short answers like FE0164010064   and FE034480001400D3 are from the coordinator, not coming from the inverter. When there is a part that contains 44810000 then it is data from the inverter.
[here](https://github.com/patience4711/ESP32-read-APS-inverters/blob/main/ESP32_ECU_v1_2.ino.bin) is the new compile.

EricMicPro, 2025-07-17:
@patience4711, Sorry I cannot make trials for now as I'm not on site where are my solar panels. 
For the other questions, as picture is always better to avoid confusion, see below:
- I used the RPI-ECU and a DS3-L
- I added the polling request frame of my INV0 (Comparison can be done between polling request and MaxPower setting request)
You can see the confirmation that short address is reversed for the polling request frame.

Just 2 remarks:
- On the polling request. It seems that the last byte of the data is the sum of all the others data bytes. Maybe it is the same for the frame to set the MaxPower ?
- The option value is set to 0x05 but this value is not in line with the bitmask from datasheet:
  0x10 = AF_ACK_REQUEST – set this bit to request APS acknowledgement for this packet
  0x20 = AF_DISCV_ROUTE – set this bit to force route discovery if a routing table entry doesn’t exist
  0x40 = AF_EN_SECURITY – set this bit to enable APS security for this packet.
As you can see on the picture, we are not using any option for the polling request (Option = 0x00).

[image]

patience4711, 2025-07-17:
@seby90 What you send via the console (and worked) should be the same as the built-in command can you please confirm that?

@EricMicPro Ahhh, that is correct. What you see in the Raspberry is already reversed. In the time i choosed to do that because it had some advantages.  In the ESP you see the not-reversed invID and as such it is used in the pollcommand. Too bad that you can't test, that would really help.

patience4711, 2025-07-17:
@kadzsol  regarding the commands you posted:
Say i have a poll command that works:
 24013A101414060001000F13 80971B01B3D6 FBFB06BB000000000000C1FEFE

and i replace the part beginning with FBFB with FBFB06C2000000000000C8FEFE i would have a zb_shutdown_single command? And that shuts-down the inverter? For how long?

EricMicPro, 2025-07-17:
@kadzsol, I saw your function command: **zb_query_inverter_info**
Do you have the explanation of the answer ?
Is there the firmware version inside ?

patience4711, 2025-07-18:
@EricMicPro You are probably right about the data length byte, in the poll command we have the same amount of data bytes end the Lenth byte = 13, not 14. 
Aynway one of both is wrong, but the poll command works, and the throttle command too, according to @seby90 
So it seems not very critical but that surprises me.

mi-hol, 2025-07-18:
> All inverter models that can be paired with an ECU-C can be throttled.

@seby90 is there maybe a document explaining this in detail? 
I recently got an ECU-R that had a "zero-feed menue" that was not mentioned in the official APsystem documentation at that time

mi-hol, 2025-07-21:
> Could you please contact me privately via the Discord address published in my GitHub profile (note: this is the first time I try this approach, if I don't reply within 48h, please comment here)

@patience4711 I've not received a private contact request yet. Did you already try?

patience4711, 2025-07-21:
In the meantime someone was kind enough to test the throttle commands (DS3) and confirmed that it acually works. And there is a response of 4 messages, the last one looks like: 

FE274481000006018A8E1414008D00D7D049000013704000204599FBFB06DE02000000000000FEFE8A8E0E2D

Sadly i have no idea what this meanse, someone?  I can recognize the inv, shortID but that's it.

Anyway, when the command fails we dont get this response so i think i can savely use this as a confirmation.

I wonder, could it be that an inverter  has a throttle value factory-installed, not being its maximum output? This could have some legal reason, f.i. to allow it to be "plug-and play" it cannot have more than xxxW in some countries.

patience4711, 2025-07-21:
Thanks, well since the command was successful it think it means passed. To my knowledge we don't get a respose on fail but i am not sure.

EricMicPro, 2025-07-21:
@kadzsol, So far few interpretation of the **zb_query_inverter_info** (Crosscheck between your YC600 and my DS3-L)

408000066781 = Inverter serial number (408 = YC600)
fbfb = Tag Start
09 = Data length (0x09 for YC600 and 0x0A for DS3-L)
dc0800500200440000 = ? (Status, FW version or ?). It is DDDC2000050000CB0000 for my DS3-L(Begin of 2023 version).
0000 = Should be the data checksum (Confirmed for DS3 but seems not done by YC600)
fefe = Tag End
xxxx = Inverter Short address (Not provided in your message)
xx = ? (Status ?) (Not provided in your message)
xx = FCS (Frame Check Sequence: XOR of all bytes between SOF and FCS) (Not provided in your message)

Regarding the disasembled, do you know how to get the FW binary file ? Or who provided the disasembled file ?
Should be interresting to get the latest FW file in order to disasemble it and look into.

kadzsol, 2025-07-21:
> [@kadzsol](https://github.com/kadzsol), So far few interpretation of the **zb_query_inverter_info** (Crosscheck between your YC600 and my DS3-L)
>
> 408000066781 = Inverter serial number (408 = YC600) [...]

Do you know your FW version? Unfortunately, I do not have an ECU so have no idea what I have on my YC600.

The firmware is coming from this guy: https://community.home-assistant.io/t/apsystems-aps-ecu-r-local-inverters-data-pull/260835/238

I have kept the old (2021) version and can give it to you. However, I am reluctant to upload it here. You can disassamble it with Ghidra.

patience4711, 2025-07-21:
@EricMicPro If you look in the issue testing throttle commands, you'l find this info on several throttle commands.

patience4711, 2025-07-21:
@mi-hol  sorry i had troubles to get on discord and i gave it up. Tomorrow i'l try it again and hope to see you there.

patience4711, 2025-07-22:
@mi-hol  The link to discord in your github profile brings me to discord but than i can't do anything, i opened a server Patience4711 and tried to send a friendshipsrequest, doesn't know mi-hol.

EricMicPro, 2025-07-22:
@kadzsol, Unfortunately same as you, I don't have any ECU so I don't know the FW version of my DS3.
Thanks for the link about the firmware.
Regarding the old one from 2021, I understand you are reluctant but how do you want to proceed ?

kadzsol, 2025-07-23:
> [@kadzsol](https://github.com/kadzsol), maybe you can use the website transfer.zip ? and then send me the link. You need to let the tab open until I download the file and it's free.

OK, but how can I send you the link?

kadzsol, 2025-07-24:
> [@kadzsol](https://github.com/kadzsol), I think you can give me the link in this chat. As soon as I will had downloaded it, I let you know then you close the tab and the file will disappear. We must just be synchonized together few minutes.

https://transfer.zip/quick#B6_wSvWznmnhiqksHnAQOqk-iFNBKW-aCb1ZwXSSH44,2c8e8aca,R

patience4711, 2025-07-26:
@EricMicPro YC600 i sent my pollcommand until FBFB and then tailed with your command
1st command sending 24013A101414060001000F1380971B01B3D6FBFB06DD000000000000E3FEFE
gives me: FE0164010064FE034480001401D2FE0345C43A1000A8FE604481000006013A1014140076007A5F0900004C408000158215FBFB3FDBFF00021003BF057F03CF0544024A059410F4470E883C0FE5020EF5A800020002000500050005000500100010000F000F0BB80BB8024A05BF0002000200000000FEFE3A100E33

the second: 24013A101414060001000F1380971B01B3D6FBFB06DE000000000000E4FEFE
gives FE0164010064FE034480001401D2FE0345C43A1000A8FE6E4481000006013A101414007300C9AB1700005A408000158215FBFB4DDE041105440FE5020F32B003CF05440FE5020F32B0066604CC0EA3A804D70214050C100FD80ED07A0F32B0056A054F019000641F3FE480068ACE8ACE000130103030190604001D241B3B6600000000FEFE3A100EAE

Is there among the possible commands a throttle command for the yc600? Because i can't get it to work, i tried with the pollcmd prefix but notting helps, i only get  the goodold FE0164010064FE034480001400D3

EricMicPro, 2025-07-27:
@patience4711, thanks for the feedback.
The YC600 replies are different than DS3.
The datalen for YC600 on first command is 63 when it is 92 for DS3.
The datalen for YC600 on second command is 77 when it is 92 for DS3.
Anyway, I did some assumptions and seems your MaxPower is set roughly to 490W maximum.
Could you confirm me that my assumptions are right or not ?
Regarding a throttle command for the yc600, let me some time to analyze the code.
I'll come back to you when I found something.

EricMicPro, 2025-07-27:
Hello @Wadzio, I'm contacting you as I saw that you shared the disassembled firmware of the ECU-C V1_1_8 several years ago.
Unfortunately, there is nothing regarding the DS3 in this version (Only YC600, YC1000 and QS1200).
So my question is: Do you have a newer version of the ECU-C firmware (Or even several versions) ?
This would be very helpful if you can share it.
Thanks in advance for your time

patience4711, 2025-07-28:
@EricMicPro  Hard to say, my power peak this year is 600W  (inverter total)  but my pannels (360W each) lay flat so they will never reach their max power i guess.  When i bought it, i thought that there was a restriction on the maxpower to 600W. Because of that you could use them as a plug-in solar system. Below 600W there are different legal rules for a legal "power generating unit".  In NL at least, but I think that these rules are eu wide implemented. 

Somewhere i have seen that the DS3 is downwards compatible with earlier models. I don't know if this means something in this matter.

EricMicPro, 2025-07-28:
@seby90, I need your help to understand the logic of the fields parameters.
If you don't mind, could you capture the frames between your ECU and the inverter after setting the following parameters:
- Reconnection time: Try with 81, 82, 83, 84, 85 then set it back to initial value
- Under voltage (Stage 2): Same principle you increase the value by 1 volt or few volts several times
- Under frequency (Stage 2): Same principle but you decrease the value by 0.1 hertz several times
- Start AC voltage high: Try with 253, 252, 251, 250, 249
If you can do it for more parameters, it will be welcome.
So far with these trials, we should get the scales of the parameters and hopefully a rule for the position of the parameters.

EricMicPro, 2025-07-31:
@patience4711, very good, I'm sure it will help.
For sure you can open a new issue for the ecu-c commands.
Regarding the disassembling analysis, I found more than 90 commands (38 for broadcast command and 52 for single inverter).
Apparently, there is no command FBFB06AA27... in the firmware version from 2021 like the one used for the DS3 .
There are 2 functions which could be what we are looking for the YC600:
- FBFB06C4... named "setfixedpowerone" but there is only 1 byte as parameter.
- FBFB06C3... named "setlimitedpowerone" but same as previous one, only 1 byte as parameter.
So it could not be the power but maybe an index (I will have a look).

It should be very nice as well if with your ecu-c, you can provide the newest firmware version especially the one for the linux version.

mi-hol, 2025-08-22:
Are you aware that APSystems published the SunSpec Document for controling an inverter via ModBus at https://apsystems.filecloudonline.com/ui/core/index.html?mode=single&path=/SHARED/%21155DN3zvMRc93csPZtsc9Ks7htSq3UizT1S6tZdx2MESTP5c2D49e7TadcB/lvGThUbRjFfSsRGR#/

There is a specific register to control the inverter power. 

Name: WMaxLimPct 
Address: 40189
Description: Set power output to specified limit

For a zero export scenario, it can be set to 0.

mi-hol, 2025-09-04:
ECU-R Firmware R1.3.19B seems to have the required code to enable Modbus communication