patience4711/ESP32-read-APS-inverters #29: request data via http broken in v1.1 & 1.2

Yoshimura87, 2025-07-17:
Hi,

I read everyday your progression ("adding commands to throttle inverters"), I'm very enthousiasm.
Even if you are not done with it, I tried the versions 1.1 and 1.2, and with both I have an issue : when I try to request data via http I get an error, the link doesn't work anymore : http://ip_of_esp_ecu/get.Inverter?inv=0
I updated via the ESP web form (using the bin files), I don't know if it matters 
When I roll back to 0.9a there is no more issue, the http data request works again.
I use http data request to retrieve signal of the inverters, because this data is not available with MQTT.
Do you think you could add signal / polled informations in MQTT ? (I also use MQTT to retrieve the power / energy).

Thanks for your help !

patience4711, 2025-07-17:
It does work, it changed to get.Data?inverter=0.  Sorry this happend because the throttle value was added.

About the mqtt, currently there are 5 formats, some of them give quite a bit of data. What are you missing, did you mean the signal strength?

Yoshimura87, 2025-07-17:
Oh thanks ! I searched in the Wiki but may be it was not updated, thanks for the information.
Yes, I monitor the signal strength, and when I created my ESP32-ECU, I think the only way to retrieve the signal strength was via http request (I use MQTT format 3, which is perfect for me).

patience4711, 2025-07-17:
according to what you get you must have changed something else, i see an array of inverters.

Anyway i added sigQ and polled in format 3. In the next release you'l see it.

closed 2025-07-17 by patience4711