patience4711/ESP32-read-APS-inverters #32: MQTT ClientID

IngoX76, 2025-07-18:
Hi everyone, I updated to version 1.2 today, and now I'm having the same problem again: the client ID isn't being saved in Mosquitto. After every restart, I have to re-enter the client ID. Furthermore, I can't do an OTA. I always connect the ESP32 to the PC and then flash it with a flash tool.

patience4711, 2025-07-18:
Correct, there is a bug causing the client_id not being saved. I had the plan to make that non configurable value like ESP32ECU-<ESP.chipid> this way we can save us some program space and its always a unique name.
Can you live with that?

IngoX76, 2025-07-18:
Thanks for the info, I think I can live with that

patience4711, 2025-07-19:
Regarding the OTA, i use it for every compile i make (so a lot) and it never fails. So i cant solve that. Maybe you flash it with a wrong memory configuration? I use minimal spiffs (1,9Mb APP with OTA /190KB spiffs)

IngoX76, 2025-07-22:
I've now got it working; I used the espHomeFlasher to flash it. 
One more question. Are the energy and power values all DC or AC values?

patience4711, 2025-07-22:
@IngoX76 Do you mean the OTA is working now? The cliend id is oke?

Yes, well to be honest i never wondered about that. All the values are reported by the inverter per channel so they must be related to the dc part  "what is generated" and not "what is the output". But that should not differ much ( i hope ).

IngoX76, 2025-07-22:
Yes, OTA works now, the client ID is OK.
Thank you very much for that.

closed 2025-07-22 by IngoX76