patience4711/ESP32-read-APS-inverters #40: Should this also work for QT2 inverters?

Hamamatsu, 2025-08-30:
Hello, I am Ton and I have paired my ESP32 and CC2530 to QT2 inverters. The result I get back are not the results that I expect.
For example the temperature of one of the inverters is 12927 degrees.
Is this project not suitable for QT2 inverters?

Regards, Ton

Hamamatsu, 2025-09-03:
Hello Patience4711, i have made a screenshot of one of my inverters. Maybe you can do something with it (or in the future). My knowledge is to little. But maybe I can help.

[image]

closed 2025-09-03 by Hamamatsu

TobiasTTM, 2025-11-01:
Hi. I have started working on adding support for QT2 inverters. See fork https://github.com/TobiasTTM/ESP32-read-APS-inverters/tree/qt2-inverter

Right now it can receive and pars the data correctly but throttle do not work yet.