patience4711/ESP32-read-APS-inverters #41: POLL request by MQTT dosn't works

domo86, 2025-09-04:
Hello, since v1.4 it is no more possible to poll by MQTT ( it works fine in v1.4c) Is it normal ?

{"poll":1} for example. No feedback by MQTT

patience4711, 2025-09-05:
As far as i can see, the mosquitto command is correctly processed. I really can't imagine why it wouldn't work. However 20 is the minimum value so that command my not work. If you can throttle via the other means it should work.
I can't test so you have to debug yourself, i'm afraid.
The first thing to do is check all your settings,  change a value in a form, change it back and save, to be sure that the configuration files are not corrupted.

Now if you open the console and send the mosquitto command you can see what is happening. Copy the 3 zigbee commands to a file. Maybe you can see the error instantly.

If you open another browser tab and throttle the inverter via the details form, you can see the same info in the console in the other tab. If you compare that with what you have in the file maybe you can see whats wrong.

For yet another way to debug [look here in the wiki](https://github.com/patience4711/ESP32-read-APS-inverters/wiki/09-TESTING-AND-DEBUGGING#api-with-debug-option)

domo86, 2025-09-06:
Hello. Thanks.

If I sends the below MQTT messages I see that the module receive date but dosn't save the value I sent :
{"throttle":2,"val":100 } => Web interface I can see 100
{"throttle":2,"val":200 } => Web interface I can see 200
{"throttle":2,"val":420 } => Web interface I can see 419 ( max ? )
{"throttle":2,"val":500 } => Web interface I can see 419 ( value is not changed )
and this for all inverters.

The problem seems to be in the decoding of after (not in the MQTT message )

This is the console feedback when {"throttle":0,"val":420 } :

* * connection opened * *
Deserialized JSON:
mqtt got message {"throttle":0,"val":420}
testing actionFlag = 240
inside actionFlag: inverter is 0
sending the throttle command for DS3
The raw powCommand for DS3 = 240182451414060001000F1380971B01A3D8FBFB06AA270000A34E01C8FEFE
sendZB FE1D240182451414060001000F1380971B01A3D8FBFB06AA270000A34E01C8FEFE3D
the raw dummy command = 240182451414060001000F1380971B01A3D8FBFB06DE00000000000000FEFE
sendZB FE1D240182451414060001000F1380971B01A3D8FBFB06DE00000000000000FEFE4A
the raw query command = 240182451414060001000F1380971B01A3D8FBFB06DE000000000000E4FEFE
sendZB FE1D240182451414060001000F1380971B01A3D8FBFB06DE000000000000E4FEFEAE
decoding inverter 0
readZB FE0164010064FE034480001401D2FE0345C482450045FE0345C482450045FE7D44810000060182451414005E003B57C1000069705000190275FBFB5CDDDE0104A34E0013BA14B413EC000A032000500003DD03A403200003E80000000000000003DD03A503350304012C060D03FF045F0E52140E1C00170258001374136F125C0014032007D023A6031401BF03D9FFFFFFFFFFFF2545C8FF1A5AFEFE82450069 rc=160
payload 5CDDDE0104A34E0013BA14B413EC000A032000500003DD03A403200003E80000000000000003DD03A503350304012C060D03FF045F0E52140E1C00170258001374136F125C0014032007D023A6031401BF03D9FFFFFFFFFFFF2545C8FF1A5AFEFE82450069
preferences key = maxPwr0
decoding a DS3
throttle value in DS3 = 419
the wanted throttle = 420

The behavior is working well in v1.3b
In v1.4a, the value shown in the web interface is not the one I sent. Seems that there is a max filter.

Also in the help console, I dont see thottle command ans poll too.
Is the action foer diag is 10;DIAG=1 because not clear in the help

Thanks for your help.

domo86, 2025-09-06:
Another check:
{"throttle":0,"val":30 } ==> Web interface show 29 
{"throttle":0,"val":40} ==> Web interface show 39
In any case, what ever the value received, the inverter behavior is not changed

domo86, 2025-09-06:
WHat I mean i that if I test on v1.3b all is working very well ( poll, throttle with any values from 20 )
If I send 420, the UI is showing 420 ( not 419) , the DS3 manage 420

With the v1.4a, 
If I send 420, the UI is showing 419, and the DS3 doesn't manage any new values reveived ( DS continues to power more that 420, no change at all whatever the value I send )

patience4711, 2025-09-06:
There is no difference as to the throttling process between the last and former versions. According your debug output the procedure works exactly the same.  The zigbee messages are set and the inverter answers, exactly as intented. 

Your inverter nicely answers that it is correctly programmed but does not actually throttle.  ? ? ?

This is a very strange phenomena, there is only one possibility; that the program does not correctly analyze the answer to the query. Begin next week i'l check that.

domo86, 2025-09-06:
Yes, I agree with you. Very stange ...
Don't hesitate to ask to test or to forward you debug ...
Have a nice week end.

domo86, 2025-09-07:
> I checked the code with ai and with an input of 420 it works oke. i can see is that your throttle command has a throttle value much too high, that cannot be programmed in the inverter. This is likely due to a corrupted calibration value. That is why i asked you to check all your values. Please set the calibration to 0.
> 
> > The first thing to do is check all your settings, change a value in a form, change it back and save, to be sure that the configuration files are not corrupted.
> 
> I a am right the problem should be gone.

Hello, 
I check all the value, and the same confirguration works fine with v1.3b.  Thottle = 500 works fine in v1.3b for the module and the inverters. I don't understand your feeback because when I dowgrade from 1.4a to 1.3b, I didn't change the configuration and all is oke.

patience4711, 2025-09-08:
That is not a conclusion. We all want to know; Does it work or does it not?

domo86, 2025-09-08:
Hello, It doesn't works. The issue is the same.
I changed values and saved. But no improvement of the behavior.
Am I alone with this issue ?
Thanks

domo86, 2025-09-09:
I can not test this evening because my inverter is sleeping ( no answer)
But I can see that I have not the same configuration panel. See below mine :

[image]

The file I used was the one you gave me in the conversation above
"No this is not normal, there was a typo somewhere in the program. [in v1_4a](https://github.com/patience4711/ESP32-read-APS-inverters/blob/main/ESP32_ECU_v1_4a.bin) it should work again."
I sure it is v1.4a because it is visible in system menu

I upload now v1.4aa

[image]

What is "calir" value ? Is is equal to 500 for my 3 inverters 

I will test tomorrow when inverters will wakeup and let you know

domo86, 2025-09-13:
Looks correct now with Trottle calib = 0 and on v1.4b
I don't know if it is again from my side, but I have to request Poll 2 or 3 time to have a correct power
For example, I know that I have 130w per DS3.
I launch MQTT poll => I receive 53
I launch MQTT poll after 2s=> I receive 126
I launch MQTT poll after 2s=> I receive 130
and this on the 3 DS3

patience4711, 2025-09-13:
When you poll once in 5 minutes you won't see these difference, it has to do with how power is calculated, see wiki

closed 2025-09-15 by patience4711